laurb9/StepperDriver #1: acceleration / deceleration

laurb9, 2016-03-06:
The stepper motor can skip steps at high load if we apply constant step frequency, but is able to go faster more reliably if gradually accelerated to target rpm. Similarly, slowly braking can help the motor stop on the target instead of missing it due to load inertia.

ben-hewitt, 2017-01-30:
Did you ever make any progress on sorting acceleration / deceleration? be suuuper useful!

Heckthor, 2017-04-21:
Any update on this?

laurb9, 2017-05-21:
I've been working on other things, but I might take a stab at it eventually. In the meantime, there is an AccelStepper library bundled with Arduino which can do this (it only needs the STEP pin).

The reference documentation for this feature would be the links below. I believe with the recent more powerful MCUs, there may be no need for formula approximations, however.
- http://www.embedded.com/design/mcus-processors-and-socs/4006438/Generate-stepper-motor-speed-profiles-in-real-time
- http://www.ti.com/lit/an/slyt482/slyt482.pdf

closed 2017-06-15 by laurb9