laurb9/StepperDriver #2: dual-motor driver

laurb9, 2016-03-06:
(investigate)

See if two motors can be moved to their respective targets at the same time, for a smooth two-axis rotation. Since motors have to draw current anyway even when not moving, power-wise this would not be a problem, but different rpms can make the timing interlacing difficult to implement.

gohilurvish, 2016-10-19:
If implemented, it would be great for CNC users.

laurb9, 2017-05-21:
I think this is relatively simple to do, although there will need to be a container class to do the management of multiple motors.
Need to see how to integrate accel/decel (#1) into this though, which might be trickier.

laurb9, 2017-05-29:
Added two new classes for multi-motor drivers:
- MultiDriver drives each motor at its standard speed, so the motors finish at different times.
- SyncDriver where all motors are synced to reach their target at the same time.

I've not yet frozen the semantics of the calls, because #19 might result in some changes there.

closed 2017-06-08 by laurb9

laurb9, 2017-06-15:
Note that multi-motor control only works well with CONSTANT_SPEED mode (#1)
MultiDriver *should* work with LINEAR_SPEED, but SyncDriver does not until I figure out the formulas to use for stretching the time.