laurb9/StepperDriver #4: esp8266 reset issue on this driver

Shaheers, 2016-06-04:
hello, on including this driver, esp8266 will repeatedly reset, it seems to take long time on execution, is any one use in their yield(); or delay(); function in files where its stuck for a while to prevent further execution so it may not rest on watch dog

laurb9, 2016-06-04:
Does this happen with the examples ?

Shaheers, 2016-06-05:
yes, i use example as is , the name of example is 'clock stepper' but its works on Arduino Uno, 
i am using esp version 2.0, IDE version 1.6.5

laurb9, 2016-06-06:
Try the branch bugfix/esp8266-reset
Also you have to change pin numbers - use D8, D9 and so on instead of 8, 9 etc.

Shaheers, 2016-06-08:
the fix use in bugfix/esp8266-reset doesn't give enough pulse to step the motor,

laurb9, 2016-06-09:
Can you clarify that ? Are you referring to timing or something else ? The fix should not have altered the pulse timing, but it maybe there's a logic level mismatch ? the ESP is 3.3V logic.

closed 2016-07-24 by laurb9