laurb9/StepperDriver #5: enhancement: enable / sleep / reset controls

meguia, 2016-06-08:
Sometimes it is desirable to switch off the motor when there is no need to maintain a holding torque. This can be incorporated in the Basic setup as a single in control that go HIGH for enabling and LOW for disabling the stepper.

laurb9, 2016-06-09:
Something like this was in the early code but I removed it in commit e1f6618. I thought the driver should not control this line anyway (it should be done externally), so other than wrapping the pin in a set of two methods there is little advantage to having it in the class and require another pin and initializer set. But I'm willing to reconsider.

Also, these drivers have two lines for switching off the motor, ~ENBL and ~SLEEP. One is active LOW the other is active HIGH. While most schematics wire ~SLEEP only, I actually prefer ~ENBL because it does not lose the counter position and does not have a charge wait.

feschiessl, 2016-06-20:
Thanks for you well-written project. I would actually like to have the enable/sleep functionality as well. It is just convenient to have the additional class function. For my project I did add the function to sent the motor to sleep...

laurb9, 2016-07-02:
Resolved in #6

closed 2016-07-02 by laurb9

laurb9, 2016-07-24:
One thought about multi-motor setups such as 2-3 axis controls: the ~ENABLE (or ~SLEEP) control line is shared, so it technically doesn't belong to any particular driver.