laurb9/StepperDriver #7: DRV8825 is using wrong ms_table[]

ZuljinSBK, 2016-06-28:
It is using ms_table[] that is defined in A4988.cpp.
It is using this table, because implementation of method setMicrostep() is in A4988.cpp and this table is static.
If I copy whole setMicrostep() method to DRV8825.cpp and that way overwrite A4988 method it start to work.
There is few method to resolve this issue like removing static and moving initialization of ms_table to constructor or overwriting setMicrostep() in DRV8825.

Additionally I found few smaller cosmetic issues :)
1) In few places there is this kind of if statment
if (!IS_CONNECTED(**ms1_pin**) || !IS_CONNECTED(**ms1_pin**) || !IS_CONNECTED(**ms1_pin**))
I think it shouldn't have only ms1_pin there.
2) Following members are not used in DRV8825.h:
    int mode0_pin = PIN_UNCONNECTED;
    int mode1_pin = PIN_UNCONNECTED;
    int mode2_pin = PIN_UNCONNECTED;

laurb9, 2016-07-02:
You're right. I'll have to brush up on my C++ to find an elegant solution to this, but it looks like wrapping the table in a getter method might be the way to do it.

hosadoya, 2016-07-11:
I have locally fixed this issue, but can't push to a remote git repo for Push Request with this error:
 Permission to laurb9/StepperDriver.git denied to hosadoya.
Attached is the fix files, can you please check-it in?
[src.zip](https://github.com/laurb9/StepperDriver/files/356462/src.zip)

laurb9, 2016-07-15:
Thanks, but I decided a shorter change would be to have a _setMicrostep(microsteps, table, size) in A4988, it looks like it should work but haven't tested it yet

laurb9, 2016-07-18:
Hmm. While this did fix the problem, there's a deeper change that needs to be made, as max_microstep isn't respected in any subclass.

laurb9, 2016-07-18:
I pushed another branch with a modified version of @hosadoya's changes which addresses the max_microstep issue, but is quite verbose.

closed 2016-07-20 by laurb9