laurb9/StepperDriver #25: Cancelling the current move

Dogmans, 2017-09-25:
Currently it does not look like there is a way of stopping a move once it has been started and "resetting" steps_remaining. Is there a nice way of doing this?

A scenario I am looking at is homing an axis, where I drive it slowly back onto the endstop and then disable the motor when that is hit. It would be nice to end the move there, rather than "waiting it out" for it to go through all the delays.

I may modify and raise a pull request if my solution ends up being at all neat.

Thanks

Dogmans, 2017-09-25:
Just a thought; presumably I can call BasicStepperDriver::move(0) in an interrupt and it will basically set steps_remaining to zero in any case.

closed 2017-09-25 by Dogmans

laurb9, 2017-09-25:
That works; you can also use non-blocking mode as shown in https://github.com/laurb9/StepperDriver/blob/master/examples/AccelTest/AccelTest.ino (you can use constant speed mode with it as well).

Dogmans, 2017-09-26:
Thanks - I've switched to non-blocking and will see how I get on.