laurb9/StepperDriver #32: this is not non-blocking

megablue, 2017-10-16:
it blocks, and worse of all it doesn't handle steps correctly. 
you can only step in full step despite the microsteps...

laurb9, 2017-10-17:
The move()/rotate() calls are blocking. Nonblocking use is shown in [the NonBlocking example sketch](https://github.com/laurb9/StepperDriver/blob/master/examples/NonBlocking/NonBlocking.ino).

About the full step vs microsteps issue, I need more information to understand the problem. If the motor is moving in full steps, it could be the driver class and driver board are mismatch, the microstep control wires aren't correct or there is an error in how they are being set. If you set the microstep connections manually, does it work ?

Perhaps if you could post a sample of your sketch, I could help more.

laurb9, 2017-10-22:
Closing, assumed resolved. 
Feel free to reopen if new information becomes available.

closed 2017-10-22 by laurb9

RobbinBaauw, 2018-01-13:
Hello, I'm also trying to understand the non-blocking sketch. I use the [Blynk](http://docs.blynk.cc/#blynk-firmware-blynktimer) library for communication with a server and lots of timers so that nothing blocks the loop. 

At the moment I solve the blocking problem by making the stepper part of the code run in a timer every 100 milliseconds, and rotating the exact degrees taking 100 milliseconds, then performing the rest of the code and then doing that all over again. It used to work, unfortunately the rest of the program got a bit more complicated and the time it takes to perform calculations in between has increased, so you will see some stuttering, which if unpreferable. Also, serial logging is a no go when using a stepper since it takes up quite a bit of time, which would make the stuttering even worse.

So now I'm trying to decide whether the non-blocking sketch would solve these problems. If I understand correctly you set like a target first (the _startMove_ function), then perform a _stop_ function when the other code needs running (for me that would be every 100-300 milliseconds, a timer would then run the stepper.stop() function). Then you calculate the time you have to perform calculations, if this time is smaller then or equal to zero it means the motor has reached its goal and you can perform all the calculations you want. If the time left is bigger than a pre defined value (in the example 100 milliseconds) you would also run your other code. 

I don't know whether this is how it works, but if it works like, what happens in the time of the _stepper.nextAction()_? Does the stepper not do anything which would provide me with the stuttering I already have or does the stepper still move and requires new data after that period of time? 

If the stuttering can be resolved, I would be really happy. Figuring out how to combine this with the rest of my code is a second thing (timers need to run in the loop, perhaps the if statement with the _wait_time_micros > 100_ in there before calling the run function of the timer) but I'm happy to work on that if this solves the issues I'm currently having. 

Thanks in advance!