laurb9/StepperDriver #33: Increasing microstep number qeually increases RPM

TechnoJLL, 2017-10-26:
RPM value is correct at full step mode, but  the number of degrees asked and the speed are multiplied by the microsteps value, hence directly going against the commented affirmation
 "  * The motor should rotate just as fast (set RPM)"

I directly implemented the MicroStepping example through a DRV8825.

am I doing something wrong? Where can I modify this?

laurb9, 2017-10-27:
Have you observed a different motor speed after changing the microsteps ? If so, then the microstep didn't get set correctly, check the wiring to M0-M1-M2 or try a different microstep.

The RPM should not change at all. If you are referring to [this line](https://github.com/laurb9/StepperDriver/blob/master/examples/MicroStepping/MicroStepping.ino#L90) in the example, please note that move() needs a step adjustment because it counts actual pulses, so when in microstepping mode the number of steps needs to change. 

The number of degrees in rotate() does not need to change at all: rotate(-180), rotate(180).

laurb9, 2017-11-01:
Closing, assumed resolved.
Feel free to reopen if new information becomes available.

closed 2017-11-01 by laurb9