laurb9/StepperDriver #40: Need higher precision for motor RPM

drogge, 2017-12-06:
Hi Laurentiu, I think you need to use something like floats for the motor RPM. I found some large errors where SyncDriver quantizes the value when it tries to adjust the motor speeds so all the steppers finish at the same time. I've seen cases where a move of around 40,000 steps can have one motor finish up .5 seconds after the other. If you like I can come up with a sample case that shows the error.

Drew

laurb9, 2017-12-06:
Thank you, I think you may be right. This is the second use case for float RPM that I've now heard of, so it might be worth doing. I don't know the impact of switching to floating point math from integer though, especially for the 8-bit chips.

You should know there is a bug in the latest release (1.1.3, see #39) resulting in incorrect speed ratios because the number of steps wasn't taken into account. You may want to try the develop branch which has a fix for that, to see if it behaves any differently, maybe the error is reduced.

Finally, if you are using acceleration, you should know that SyncDriver code just "wings it" there so it will likely be imprecise in that mode.

drogge, 2017-12-06:
Thanks for the reply. I'd already found and fixed the problem the speed ratio stuff so that's not part of the problem. As far as I can tell, the rpm value is only used at the start of a move or when calculating the time a move will take so going to float shouldn't have much of a realtime performance hit. What's the other use case for a float RPM? Finally thanks for the warning for about SyncDriver.

laurb9, 2017-12-08:
Hmm, you're right, I was thinking that if we switch rpm to float then the rest of the math would have to follow, but maybe it doesn't have to. The only other thing that would happen is that pulling in all the fp math lib would add another 1K or so to the code on AVR, I am not sure if I should be too concerned about that.

The other use case for float RPM is a precision syringe pump that needs to deliver a precise volume/s.

I'm thinking a [fake] gear ratio could serve the same purpose, but it's not as clean.

stefanlechner, 2018-01-28:
Well I changed the lib to use floats for RPM but is still don't produce the desired results.
I'ts coming better but I'vs still a gap of about .2s in my case.
The only way to solve the problem, seems to me, calculating additional ys to the pulses.
In my case I would have to add a additional ys on every third pulse.

laurb9, 2018-01-29:
Probably all the rest of the math has to be done in float, otherwise the integer quantization of the micros could account for that error.

stefanlechner, 2018-01-29:
I did a lot of experiments, last weekend. Float RPM will not solve the problem. The calculation across RPM itself seems to be the problem. RPM = Rotations per MINUTE. We are calculating step pulses in µs. In my case I've a ratio of 12.85 between the axis. Actually a startMove(259200,20160) should end within the window of the short pulse, to not have a stop for the next move. 
For now, the only way I can imaging is to use RPM for the time estimate, then we have to operate on steps and time in µs.
Btw. your RPM are depending on microsteps. Imho this shuld'n be. RPM is a fixed definition so I have to half the pulse when doubling the microsteps and not doubling the RPM to get the same speed.
I did a fork, and will try to redo the syncdriver.

laurb9, 2019-02-09:
For now I implemented the float RPM only, which is pretty useful in and of its own. For the other cases, increasing the resolution of the micros calculation could help with drift, but I'll address this later if it becomes a problem

closed 2019-02-09 by laurb9