laurb9/StepperDriver #42: Update version on Arduino repos

floriansegginger, 2017-12-22:
First of all: great library! Thanks!

Right now, the version available through the Arduino.cc IDE is 1.1.2, which has some unpatched bugs (most notably #36 and #28).

laurb9, 2018-01-21:
I didn't know this is available through the IDE. I will check.

per1234, 2018-02-06:
@floriansegginger are you talking about the Arduino IDE Library Manager (Sketch > Include Library > Manage Libraries)? I don't find any library named StepperDriver in the [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json) and I also don't find any [request ](https://github.com/arduino/Arduino/issues?q=is%3Aissue+StepperDriver)for this library to have been added.

Note that Library Manager shows all libraries you have installed in addition to the ones available for installation from the index. So if you had previously manually installed StepperDriver 1.1.2 it would appear in Library Manager and not show the option to update to a different version.

@laurb9 if you are interested in adding your library to the Library Manager index it's quite easy, you only need to submit a request to the [arduino/Arduino issue tracker](https://github.com/arduino/Arduino/issues). Your library already meets all the requirements. Details here:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

laurb9, 2019-01-30:
Opened https://github.com/arduino/Arduino/issues/8462 to publish the StepperDriver lib to Arduino Library manager.

closed 2019-01-30 by laurb9