laurb9/StepperDriver #43: Non-blocking issue

RobbinBaauw, 2018-01-13:
Hello, I posted a comment on a [closed issue](https://github.com/laurb9/StepperDriver/issues/32) and didn't see it appearing anywhere, so I'm copying it here and removing the comment in that issue. Anyway, I was also trying to understand the non-blocking sketch. I use the [Blynk](http://docs.blynk.cc/#blynk-firmware-blynktimer) library for communication with a server and lots of timers so that nothing blocks the loop. 

At the moment I solve the blocking problem by making the stepper part of the code run in a timer every 100 milliseconds, and rotating the exact degrees taking 100 milliseconds, then performing the rest of the code and then doing that all over again. It used to work, unfortunately the rest of the program got a bit more complicated and the time it takes to perform calculations in between has increased, so you will see some stuttering, which if unpreferable. Also, serial logging is a no go when using a stepper since it takes up quite a bit of time, which would make the stuttering even worse.

So now I'm trying to decide whether the non-blocking sketch would solve these problems. If I understand correctly you set like a target first (the _startMove_ function), then perform a _stop_ function when the other code needs running (for me that would be every 100-300 milliseconds, a timer would then run the stepper.stop() function). Then you calculate the time you have to perform calculations, if this time is smaller then or equal to zero it means the motor has reached its goal and you can perform all the calculations you want. If the time left is bigger than a pre defined value (in the example 100 milliseconds) you would also run your other code. 

I don't know whether this is how it works, but if it works like, what happens in the time of the _stepper.nextAction()_? Does the stepper not do anything which would provide me with the stuttering I already have or does the stepper still move and requires new data after that period of time? 

If the stuttering can be resolved, I would be really happy. Figuring out how to combine this with the rest of my code is a second thing (timers need to run in the loop, perhaps the if statement with the _wait_time_micros > 100_ in there before calling the run function of the timer) but I'm happy to work on that if this solves the issues I'm currently having. 

Thanks in advance!

laurb9, 2018-01-21:
The non-blocking implementation has three phases. First you do startMove() which precalculates the timings and sets up the step counters, then you call nextAction() periodically at the indicated intervals to trigger the pulses. Each call returns how many milliseconds you have until the next move, and you can use that to decide if you have enough time to do something else.

You do not need to call stop() unless you want to cancel the move, because it will reset all the step counters to zero. If you take too long between nextAction() calls, the timing of the pulses will be off, the motor movement will be less smooth but that's about it.

If you do need a lot of time while the motor is moving, I am thinking that you could try to use fewer microsteps to increase the step interval.

laurb9, 2018-01-21:
What I would really like to do is use interrupts for the timing, which would make the entire motor activity async, but I didn't have time to look into that and there are other libraries which do this better.

PotatoSauceVFX, 2018-08-24:
Has anyone found a good solution to this problem? I am using ESP32 AsyncWebServer and when the motor is moving it freezes and jerks for a second when requests are made. Is there another library that someone has found that uses interrupts?

laurb9, 2018-08-25:
Try taking a look at the AccelStepper library, it has been around for a longer time. 
With interrupts, maintaining the functionality across different chips becomes an issue.

RJ-Fynydd, 2018-08-25:
Thank you very much @laurb9. I'll check it out.