laurb9/StepperDriver #44: Integeroverflow using syncdriver with large values

stefanlechner, 2018-01-27:
Two motors
Motor 1  - 200 Steps 16 microsteps   14 RPM
Motor 2-  200 Steps 16 microsteps 180 RPM

Motor 1 19530 steps to go.
Motor 2  26 * 3 * 200 * 16 steps to go

Motor 1 getTimeForMove = 25994007 (yes it's already as synced as possible).
Motor 2 getTimeForMove = 25958400

when I'm going now with a
controller.startMove( 25994007,25958400)
it calculates  
Motor one RPM 14
Motor two RPM 11

maybe it would be better to calculate a float factor and apply it to the rpm value , instead of calculating with integers over the large value.

    /*
     * Stretch timing for all others by adjusting rpm proportionally
     */
    if (move_time){
        FOREACH_MOTOR(
            if (steps[i]){
                rpms[i] = motors[i]->getRPM();
                Serial.print(rpms[i]);Serial.print(";");Serial.print(timing[i]);Serial.print(";");Serial.println(move_time);
                **motors[i]->setRPM(round(rpms[i] * (float(timing[i]) / float(move_time))));**
            } else {
                rpms[i] = 0;
            }
        );
    }

But after changing there is an other thing I cant understand at all, why is the first motor adjusted even if it needs the longest time.
The printed values are quite strange (careful downto loop):
180;26956800;26956800
3840;80640;26956800

laurb9, 2018-01-29:
Can you check this also ? startMove() expects number of steps as an argument, but it looks like you are passing micros.
> when I'm going now with a
> controller.startMove( 25994007,25958400)

And yes, as you noticed in #40 I should just switch the rpm to float.

stefanlechner, 2018-01-29:
yes sorry, I copied the wrong values 
correct would be
controller.startMove(19530,249600).

But have a look on the #40.
I'll close this

closed 2018-01-29 by stefanlechner