laurb9/StepperDriver #47: adding TMC 2100 driver to the list

pedros89, 2018-02-12:
Any chance this library could be used with TMC 2100 in the future?
I like TMC 2100 as they are more silent drivers.
Pedro

pedros89, 2018-02-16:
So basically if I just substitute the A4988 that I am currently using with the TMC 2100 it should already be working without any library modification?

laurb9, 2018-02-16:
Most likely not, the control wires are different even if the carrier might be the same. A4988 has three microstep pins (MS1, MS2, MS3), the TMC2100 has two (CFG1, CFG2) which means one of the carrier pins does something else (I don't have a TMC2100 carrier to see what).

TedyLupu, 2018-06-19:
Hmmm... i'm using @ the moment the TMC 2100 and it works ok with tge a4998 lyb.  there are some problems when running the motor intermittently due to the fact that the tmc has that "interpolator" for microsteping and the pulse delays are messed up. But it works in continuous mode