laurb9/StepperDriver #48: 'ISO C++ forbids initialization of member 'mode''

eslopez92, 2018-02-19:
Hi Laurens,

I found your source code for the DRV8834 using Arduino: https://github.com/laurb9/StepperDriver

I am also running into the similar issue using Windows 7 and the Arduino 1.0.5-r2 IDE.

**In file included from /A4988.h:13,
                 from A4988.cpp:10:
BasicStepperDriver.h:36: error: ISO C++ forbids initialization of member 'mode'
BasicStepperDriver.h:36: error: making 'mode' static
BasicStepperDriver.h:36: error: ISO C++ forbids in-class initialization of non-const static member 'mode'
BasicStepperDriver.h:37: error: ISO C++ forbids initialization of member 'accel'
BasicStepperDriver.h:37: error: making 'accel' static
BasicStepperDriver.h:37: error: ISO C++ forbids in-class initialization of non-const static member 'accel'
BasicStepperDriver.h:38: error: ISO C++ forbids initialization of member 'decel'
BasicStepperDriver.h:38: error: making 'decel' static
BasicStepperDriver.h:38: error: ISO C++ forbids in-class initialization of non-const static member 'decel'
BasicStepperDriver.h:56: error: ISO C++ forbids initialization of member 'last_action_end'
BasicStepperDriver.h:56: error: making 'last_action_end' static
BasicStepperDriver.h:56: error: ISO C++ forbids in-class initialization of non-const static member 'last_action_end'
BasicStepperDriver.h:57: error: ISO C++ forbids initialization of member 'next_action_interval'
BasicStepperDriver.h:57: error: making 'next_action_interval' static
BasicStepperDriver.h:57: error: ISO C++ forbids in-class initialization of non-const static member 'next_action_interval'
BasicStepperDriver.h:70: error: ISO C++ forbids initialization of member 'enable_pin'
BasicStepperDriver.h:70: error: making 'enable_pin' static
BasicStepperDriver.h:70: error: ISO C++ forbids in-class initialization of non-const static member 'enable_pin'
BasicStepperDriver.h:74: error: ISO C++ forbids initialization of member 'microsteps'
BasicStepperDriver.h:74: error: making 'microsteps' static
BasicStepperDriver.h:74: error: ISO C++ forbids in-class initialization of non-const static member 'microsteps'
BasicStepperDriver.h:82: error: ISO C++ forbids initialization of member 'rpm'
BasicStepperDriver.h:82: error: making 'rpm' static
BasicStepperDriver.h:82: error: ISO C++ forbids in-class initialization of non-const static member 'rpm'
/BasicStepperDriver.h: In static member function 'static void BasicStepperDriver::delayMicros(long unsigned int, long unsigned int)':
BasicStepperDriver.h:46: error: 'yield' was not declared in this scope
In file included from A4988.cpp:10:
/A4988.h: At global scope:
A4988.h:18: error: ISO C++ forbids initialization of member 'ms1_pin'
A4988.h:18: error: making 'ms1_pin' static
A4988.h:18: error: ISO C++ forbids in-class initialization of non-const static member 'ms1_pin'
A4988.h:19: error: ISO C++ forbids initialization of member 'ms2_pin'
A4988.h:19: error: making 'ms2_pin' static
A4988.h:19: error: ISO C++ forbids in-class initialization of non-const static member 'ms2_pin'
A4988.h:20: error: ISO C++ forbids initialization of member 'ms3_pin'
A4988.h:20: error: making 'ms3_pin' static
A4988.h:20: error: ISO C++ forbids in-class initialization of non-const static member 'ms3_pin'
A4988.h:34: error: expected ';' before 'override'
A4988.h:54: error: expected ';' before 'override'
A4988.cpp: In constructor 'A4988::A4988(short int, short int, short int, short int, short int, short int)':
A4988.cpp:36: error: 'short int A4988::ms1_pin' is a static data member; it can only be initialized at its definition
A4988.cpp:36: error: 'short int A4988::ms2_pin' is a static data member; it can only be initialized at its definition
A4988.cpp:36: error: 'short int A4988::ms3_pin' is a static data member; it can only be initialized at its definition
A4988.cpp: In constructor 'A4988::A4988(short int, short int, short int, short int, short int, short int, short int)':
A4988.cpp:41: error: 'short int A4988::ms1_pin' is a static data member; it can only be initialized at its definition
A4988.cpp:41: error: 'short int A4988::ms2_pin' is a static data member; it can only be initialized at its definition
A4988.cpp:41: error: 'short int A4988::ms3_pin' is a static data member; it can only be initialized at its definition
A4988.cpp: At global scope:
A4988.cpp:61: error: no 'short int A4988::setMicrostep(short int)' member function declared in class 'A4988'
A4988.cpp:93: error: no 'short int A4988::getMaxMicrostep()' member function declared in class 'A4988'**

I am not sure if this is a Windows problem, or what?

Thanks!
Emilio

laurb9, 2018-02-19:
Arduino 1.0.5 is pretty old, the current version is 1.8.5; have you tried using a more recent one ?

I'm sure I'm taking some liberties with C++ but at least with 1.8.5 the code compiles fine.

laurb9, 2019-01-30:
If I move all the initializers in the constructors and make the initial values constant, this will make it possible to override some of the settings (timings, microsteps) for each driver without using virtuals.

laurb9, 2019-01-30:
Note: As of version `1.6.6`, the Arduino IDE enables `C++11` by default. 
See https://en.wikipedia.org/wiki/C%2B%2B11#Object_construction_improvement
So it is no longer necessary to copy all the member initializations to the constructor.

closed 2019-01-30 by laurb9