laurb9/StepperDriver #49: weird behavior with TB6600

gbdb, 2018-02-26:
Hello, complete noob, needs advise on his setup.
single motor, 57BYGH420-2 ( nema23, 2a per coil, 

https://french.alibaba.com/product-detail/stepper-motor-57bygh420-2-125-oz-in-200-steps-rev-600mm-wire--60392070742.html

and the driver https://www.dfrobot.com/wiki/index.php/TB6600_Stepper_Motor_Driver_SKU:_DRI0043
my dip switches are ON-ON-OFF ( so the micro step are set to 1 and code is using 1 as well 
#define MICROSTEPS 1 )
ON-OFF-OFF so it's setup to 2A as my motor spec.

wiring looks great since I;m able to run basic for loop and motor turn smoothly in both direction.
I' using the Enable pin (I'll need that later) but just trying to use the library for now..
BasicStepperDriver stepper(MOTOR_STEPS, DIR, STEP);

basic example will make the motor turn 1/8 turn and go back to original position...

sorry for asking before spending more hours on the thing but I just left the DRV8825 for this stronger controler and I hit different issues and I'm getting tired of this project lol

laurb9, 2018-02-27:
Is it possible the switches could be set backwards which would make it run at 1:16 ? Since your for-loop works, run it with exactly 200 steps, which should be a full revolution, and see what the motor does.

If it does a full turn, then the switches are okay and other thing I can think of is that the pulse timing may be too short. You could try setting step_high_min to a larger value in [BasicStepperDriver.h](https://github.com/laurb9/StepperDriver/blob/master/src/BasicStepperDriver.h#L76). I noticed they are using a common-anode configuration for the pins, but the library assumes common ground so your signals will be inverted.

gbdb, 2018-02-27:
oh. wow. I think you found something. a for loop of 200 will make the motor 'shake' 
using 3200 in the loop will make the stepper to do a single rotation.
OFF ,OFF, ON (witch is supposed to be correct since it`s 16
but even ON ON OFF....   witch is supposed to be 1 (then 200 steps will make a rotation from my understanding) will still do a single turn when using 3200 the loop !!! the switch are not even working correctly!!

using your BasicStepperDriver.ino example, setting microsteps to 16, it does work !!! *in almost any dip switch configurations.

I'm I crazy or my driver is Again the issue ?? I was running way from DRV8825 hoping to get a more reliable board but it seams not to be the case...I'll get a second driver from the local hobby shop and try to return the one I have

laurb9, 2018-03-01:
From what you are describing it sounds like it could be driver, or the microswitch groups 1-3 and 4-6 are swapped so current control and microsteps are swapped. I'd ask the company you purchased the driver from, or find a source for more reputable drivers. Pololu has a lot of them but few go past 1.8A without additional heatsinking.

gbdb, 2018-03-04:
with my second driver, this one from DF Robot. looks a little more sturdy since the board is screwed down. can you imagine the other one was held down only by the chip heat sinking screws!! ohh wow.

I get the same ~ behavior on that new one. can it possible this specific motor is only working at 16 microsteps ?? I really take my time to wire everything up and can't think of any reason I get these behavior.. should be simple to run a single motor up and down..

laurb9, 2018-03-09:
The motor does not know about microsteps, which are where the driver maintains the magnetic fields in balance to keep the rotor in a position between the real “steps”.

So I do not know what could cause it to appear to run at 16. Try to check the timing of the delays and maybe increase them to move slower, maybe you’re sending too fast and the motor can’t keep up especially if it has a load

laurb9, 2018-03-23:
Closing, assumed resolved

closed 2018-03-23 by laurb9