laurb9/StepperDriver #52: Freeze after a number of cycles

TComp, 2018-03-31:
Hi, I have an application where a stepper motor has to run every minute and always in th same direction. I am using an Arduino Uno, a PG15S-020 geared stepper motor and a Pololu D8834 driver. I have also added a relay that turns off the power to the driver/motor to save battery life and avoid motor heating. A2 pin is used as output for the relay since all other outputs are taken. Below is the simplest code I have tested:

#include <Arduino.h>
#include "BasicStepperDriver.h"

#define MOTOR_STEPS 1080
#define RPM 20
#define MICROSTEPS 1
#define DIR 1
#define STEP01 4
#define SECOND 1000L
#define MINUTE (60*SECOND)
#define HOUR (60*MINUTE)
BasicStepperDriver stepper01(MOTOR_STEPS, DIR, STEP01);

void setup(void) {
stepper01.begin(RPM, MICROSTEPS);
pinMode(A2, OUTPUT);        // Set pin A0 as output for relay
digitalWrite(A2, LOW);         // De-energize relay
}

void loop(void) {
digitalWrite(A2, HIGH);       // Energize motor power relay
delay(200);                          // Make sure relay is energized
stepper01.rotate(360);        // Run stepper  --> SKETCH HANGS RIGHT HERE
delay(750);                         // Wait until stepper is done
digitalWrite(A2, LOW);       // De-energize relay
delay(MINUTE);                  // Wait for 1 minute
}

This code always hangs at loop #69 at the stepper01.rotate command. I am testing this code without connecting the driver and stepper, only with the Arduino and the relay. If I change the delay from 1 minute to 30 seconds, the code hangs at the same command at loop #129. I changed from delay() to millis() and the code hangs always at the same command at loop #72. Am I doing something wrong here?
Thanks for the support.

laurb9, 2018-03-31:
I am unable to reproduce the issue on Arduino Uno (it is at loop 90 and still going). I am using Arduino 1.8.5 and StepperDriver 1.1.3 (please check your code with latest versions).

The exact code I ran is below (the only change I've made is to print loop number to serial). It is possible that some other part left out of the original code is causing the problem.
```C++
#include <Arduino.h>
#include "BasicStepperDriver.h"

#define MOTOR_STEPS 1080
#define RPM 20
#define MICROSTEPS 1
#define DIR 1
#define STEP01 13
#define SECOND 1000L
#define MINUTE (60*SECOND)
#define HOUR (60*MINUTE)
BasicStepperDriver stepper01(MOTOR_STEPS, DIR, STEP01);

int x = 0;

void setup(void) {
Serial.begin(115200);
stepper01.begin(RPM, MICROSTEPS);
pinMode(A2, OUTPUT); // Set pin A0 as output for relay
digitalWrite(A2, LOW); // De-energize relay
}

void loop(void) {
digitalWrite(A2, HIGH); // Energize motor power relay
delay(200); // Make sure relay is energized
Serial.println(x);
stepper01.rotate(360); // Run stepper --> SKETCH HANGS RIGHT HERE
Serial.println(x++);
delay(750); // Wait until stepper is done
digitalWrite(A2, LOW); // De-energize relay
delay(MINUTE); // Wait for 1 minute
}
```

thymjan, 2018-03-31:
Why do you use a relay? The DRV8834 offers sleep and enable pins.

TComp, 2018-03-31:
Thanks for the prompt reply. The problem was the version of the library. I was using version 1.1.2 (latest version of the library from within the 1.8.5 IDE). I had this issue consistently in 2 different Arduino Uno boards without having any pins connected to anything, just the USB cable from the PC and the board. 
After updating manually the library to the 1.1.3 version from GitHub, the problem disappeared.

Thymjan, I am using a relay because the all the other pins are taken, I have 10 steppers (10 pins for STEP), one bluetooth module (2 pins), 1 buzzer (1 pin), 1 RTC DS3231 module (2 pins) and an IIC oled screen (2 pins). The relay needs only 1 pin and I am able to toggle power ON and OFF to all 10 drivers/steppers together for the length of time I need them to spin (around 600-700 ms every minute). The code snippet I sent earlier is a test code with one stepper only just to verify the problem I was having.

In any case, thank again for taking time to test my code and helping with the tip to update the library.

laurb9, 2018-03-31:
Thanks for confirming. The issue you reported is then most likely same as #36 which was indeed fixed in 1.1.3.

laurb9, 2018-03-31:
See #36

closed 2018-03-31 by laurb9

thymjan, 2018-03-31:
Instead of the relay you could use the relay pin for all enable pins, couldn‘t you?

TComp, 2018-03-31:
You are absolutely correct. To be honest, I did not think about it at all. In any case, it would have added 10 more wires on top of what I already have and I also would have had to solder 10 more pins to the DRD8834 drivers. On top of this, I am using 3-wire plugs to connect to the steppers, like the ones that are usually used for servo motors (red/black/white) for 5VDC, ground and STEP signal. A fourth wire for the ENABLE signal would have complicated my design. I have connected all 10 5V red wires to a connector block that takes the 5VDC from the relay contacts. It may not be the most elegant solution but it works perfectly for my purpose.