laurb9/StepperDriver #55: Missed step detection

limikael, 2018-05-10:
This is not a bug report, but rather a support request. Posting it here, since I don't know where the proper place is. If it is the wrong place to ask the question, I apologize.

I would like to do missed step detection, as is done in the video on this page:

[MISSED STEP DETECTION WITH MK2CLONE R2](https://prusaclone.wordpress.com/2018/03/01/missed-step-detection-with-mk2clone-r2/)

Is this possible with this library? And is there a tutorial where I can learn how to set it up, both on the hardware and software side?

laurb9, 2018-05-11:
Missed step detection ctually relies on the driver chip doing it, by measuring the back EMF from the motor coils, so it's not a software solution. Those are more advanced chips and come with other features such as I2C or SPI control.

The currently supported chips do not have this feature. The TMC2130 referenced in your link does, maybe also AMIS-30543, neither are yet supported by this library. But the youtube video links to another library which is made specifically for TMC2130 that probably supports the feature you are looking for.

limikael, 2018-05-11:
That clears things up! Thank you very much!

closed 2018-05-11 by limikael