laurb9/StepperDriver #56: i need to use 8 step motor

nuedjidee, 2018-05-26:
i need to use 8 step motor can use this library

laurb9, 2018-05-26:
Do you have an 8-step or 8-wire motor ?

If you really have an 8 step motor, then you can set MOTOR_STEPS to 8 and it should work.

If you have an 8-wire stepper, take a look at [Adafruit's All About Stepper Motors](https://learn.adafruit.com/all-about-stepper-motors/types-of-steppers) page to learn how to wire it as a bipolar motor that the drivers can work with.

nuedjidee, 2018-05-26:
I have 8 step motor not 8 wire motor

I want to control 8 motor

How I can set this library. Please tell me

Thank you sir

nuedjidee, 2018-05-28:
this library can use 8 motor ?sir

laurb9, 2018-06-02:
8 motors ? If you want to drive them sequentially, you can have as many as you want. But in parallel, the max the library does as implemented is 3. With some code changes it can do more but I do not support that use because you can run into timing issues.

nuedjidee, 2018-06-02:
thank sir

closed 2018-06-04 by laurb9