laurb9/StepperDriver #57: count the number of steps

tharunjoy, 2018-05-30:
Is there any way to count the number of step's a stepper motor using the library.

laurb9, 2018-06-04:
do you mean the number of steps since initialization, since last move, or predicted steps for a move ?

tharunjoy, 2018-06-05:
steps the motor has moved or the completed steps of the motor

LewisMojica, 2018-12-01:
This would be a very useful feature.

tognotommy, 2019-01-23:
I would also be interested in this feature, was thinking of implementing it myself. It would be for me however tracking the performed steps since last command issued probably, since I need to compare that value with encoder readings. If i do implement it i'll let you guys know :)

laurb9, 2019-01-28:
Thank you everyone for your feedback. This method will be available in 1.1.4

closed 2019-01-28 by laurb9