laurb9/StepperDriver #58: Control motor by speed

acmilan567, 2018-06-13:
Hi, I need to control the motor according to a speed of input rather than by the number of steps. Precisely, I would like a continuous movement, of which I can however change the speed when needed. In case it was already possible, could you tell me how?

JonIvanArt, 2018-06-13:
I'm running into the same issue. Since steppers come in different revolutions per minute, some geared and some not, you have to have some type of position sensor or optocoupler to detect RPMs in order to make this happen.

acmilan567, 2018-06-17:
I need to know if there is any possibility to solve my problem

laurb9, 2018-06-17:
If you can use constant speed (no acceleration) and non-blocking mode then you can change the RPM mid-flight and keep extending the runtime by calling setRPM() followed by startMove() inside the loop to recalculate timings. 

You would have to change the RPM by small deltas or it may skip steps.

laurb9, 2018-07-11:
Closing, assumed answered

closed 2018-07-11 by laurb9