laurb9/StepperDriver #60: Question about pulse width

T-Kuhn, 2018-07-10:
I tried to use this library with a TB6600 stepper driver ( https://www.dfrobot.com/wiki/index.php/TB6600_Stepper_Motor_Driver_SKU:_DRI0043 )

It didn't move (I used the code from the BasicStepperDriver example) and I am wondering why and whether or not there is some parameter that could be tweaked in order to get it going. Here are some things I checked:
- The driver works fine when I send it pulses through some self-derived pulse-generating algorithm thingy.
- The coils of the motor are indeed energized.
- There are pulses getting sent to the driver (checked with an oscilloscope)

All this leads me to believe that maybe the pulse width is too thin for the driver to pick up. Any pointers or thoughts would be much appreciated. I will try to get it to work again this evening and update/close this issue if I am able to resolve it.

laurb9, 2018-07-10:
That's possible especially if you have a fast mcpu. Try changing step_high_min to 5 or 10 (microseconds) in https://github.com/laurb9/StepperDriver/blob/master/src/BasicStepperDriver.h#L76 and see if that changes anything.

T-Kuhn, 2018-07-11:
Those width parameters were exactly what I was looking for.
And here are some working values:
```
    static const int step_high_min = 30;
    static const int step_low_min = 30;
```
I tested 5, 10, 20, 30, 40, 50. And at 30us the thing finally started to rotate.

closed 2018-07-11 by T-Kuhn

laurb9, 2018-07-12:
I looked up the data sheet for [Toshiba TB6600HG](https://toshiba.semicon-storage.com/ap-en/product/linear/motordriver/detail.TB6600HG.html), assuming that's what you are using, and it lists a minimum pulse width of 2.2µs. So 30µs sounds really, really slow!

Does the output signal look square when connected to the driver ? A capacitive load could delay reaching the 2V required to register logic high.