laurb9/StepperDriver #62: blue led turns on and stays on after begin(). NodeMCU 12E

nospam2k, 2018-08-23:
I am loading the driver as follows:

#include <Arduino.h>
#include "DRV8825.h"

#define MOTOR_STEPS 48
#define RPM 20 //200
#define ENABLE 15
#define DIR 0
#define STEP 2
#define MODE0 13
#define MODE1 12
#define MODE2 14

DRV8825 stepper(MOTOR_STEPS, DIR, STEP, ENABLE, MODE0, MODE1, MODE2);

void setup() {
  stepper.begin(RPM);
}

void loop() {
}

When I load the sketch, as soon as it runs stepper.begin(RPM) the blue led stays on. No other stepper code is running. If I rem out stepper.begin(), blue led stays off.

laurb9, 2018-08-25:
Pin 13 is traditionally the LED pin on Arduino, so it might be on or off depending on what MODE0 needs to be at.

nospam2k, 2018-08-28:
Not pin 13 but you did give me what I needed. It is the STEP pin (D4). Thanks! I didn't realize there were two leds in the Nodemcu-12E.

closed 2018-08-28 by nospam2k