laurb9/StepperDriver #66: Teensy 3.2 and DRV8834

arasabbasi, 2018-10-08:
Hi, 

when I use a teensy 3.2 and a DRV8834, the library hangs itself at the second run of the line `delayMicros(next_action_interval, last_action_end);` in `long BasicStepperDriver::nextAction(void){`.

And if I use your library in arduino uno, it works neither.

When I use my own written library in teensy my stepper works...

So confusing. 
Any ideas?

laurb9, 2018-10-10:
It may have to do with the configuration settings, microsteps, or some specific thing to Teensy 3.2 compiler, but I have very little to speculate on because I don't see the rest of the code. 

Try to post the entire sketch, the smallest size that can still reproduce the problem. Also, you should start by using some of the examples.

laurb9, 2018-10-21:
I was using Teensy 3.1 and Teensy LC with DRV8834 in this project https://github.com/laurb9/PanoController-Firmware before I switched to Feather M0. I do not have Teensy 3.2.

I have also tested it with Arduino UNO (but not recently), A4988 driver and a few others. The timing on step/dir is common to all though, just microstep config and special features differ.

Regarding the high impedance mode, it may be a chip config issue so I'd ask in the PJRC forums.

The pins can be configured with pull-down or pull-up on input. INPUT should disable them (as opposed to INPUT_PULLUP etc). Or try to use an analog input pin, as I imagine a pull down can mess a voltage reading there.

laurb9, 2018-12-10:
Closing due to no more input, assumed resolved. Feel free to reopen if needed.

closed 2018-12-10 by laurb9