laurb9/StepperDriver #67: Wrong ENABLE polarisation for DRV8825

mudirk, 2018-10-16:
Same as #51 for the DRV8825. To enable the driver the enable pin must be pulled up (HIGH) and to disable the driver the enable pin must be pulled down (LOW)

luksa, 2018-11-21:
That's not true. The DRV8825 datasheet clearly states that ENABLE must be pulled DOWN to enable the driver.

From http://www.ti.com/lit/ds/symlink/drv8825.pdf:

> nENBL: Logic *high to disable* device outputs and indexer operation, logic *low to enable*. Internal pulldown.

andreypopov, 2018-12-05:
These default RESET and SLEEP states are ones that prevent the driver from operating; both of these pins must be high to enable the driver https://www.pololu.com/product/2133
LOW -> DISABLE
HIGH -> ENABLE

with your library enable() disables my stepper, disable() - enables..

How to fix it? should I change stepper pins ?

laurb9, 2019-01-26:
The enable function was meant for the `ENABLE` pin on the carrier board, so from that perspective it works correctly, sending a logic `LOW` to enable. `SLEEP` would be hardwired to `HIGH` in this model *).

The Pololu minimal wiring suggests controling via `SLEEP` instead and logic HIGH to enable, which causes this issue and it's a problem specific to this particular board. Since a lot of people will wire it based on the example, I think we may need a way to select one or the other instead.

*) The reason for using `ENABLE` is because while `SLEEP` does reduce consumption to a super low 10µA, it also resets the internal position counter *and* needs >1ms time to wake up and be able to read STEP inputs.

The few mA that the driver uses in disabled state are similar to MCU usage and maybe dwarfed by the first few milliseconds of stepper motion anyway.