laurb9/StepperDriver #71: SyncDriver not working

paulelsner, 2018-12-06:
Hi,
I choose your library for the sync driver feature but it does not seem to do what is should. My motors stop turning at different times.
My setup is 3 motors with 64 steps configured and an RPM of 500 on all of them.

I tested the move and the startMove function. Neither is working as expected.
Thanks!

laurb9, 2018-12-10:
SyncDriver does not work with accelerated mode because I haven't done the calculations needed. I'm guessing you are using this mode because of the 500 RPM ?

paulelsner, 2018-12-10:
What do you mean with accelerated mode?
I only call begin(500) on the driver. Actually I had to go up to 1500 with my motors to get the speed I want.

laurb9, 2018-12-27:
Accelerated mode is where the motor ramps up to speed smoothly as opposed to starting at max rpm right away. You can tell because something like this would be in the code: `stepper.setSpeedProfile(stepper.LINEAR_SPEED, MOTOR_ACCEL, MOTOR_DECEL);`.

Please try the MultiAxis example with two motors and see if that works. If it does, then add another motor and see if that still works. Or if you already have a simple sketch please attach it here and it would also help to know what type of board you are using.

Are the motors at least reaching their target position even if at different times ? or are they stopping before they are supposed to ? Does it work with shorter moves ?

laurb9, 2018-12-27:
Ah. I think I know what this could be. I fixed a timing issue (#39) a while ago but didn't make a new release so it's not fixed in the "official" version. Try the `develop` branch if you can, or make this small change to the code: https://github.com/laurb9/StepperDriver/commit/c8ac6ff6259a506bda59eab430071d0b84ab260d

laurb9, 2019-01-27:
Assuming that `develop` branch fixed the issue reported, please reopen if that was not the case.

closed 2019-01-27 by laurb9