laurb9/StepperDriver #76: Non blocking constant rotation

jonathan-moss, 2019-01-27:
Is there a way to set the motor to a constant rotation (with a set speed) until .stop() is called, without blocking the rest of the program.

I have been looking at the examples and reading through the code but I can't seem to make this work.

I did manage to call 
Setup > stepper.startMove(3600000);
Loop > stepper.nextAction(); 

But the motor turns very slowly or sometimes not at all.

Thanks.

Great driver, and thank you for all the work you have put in to this.

laurb9, 2019-01-28:
That is how it should work, as in the `NonBlocking` example sketch.
Try to see if the standard examples `BasicStepperDriver` and `MicroStepping` are working with a free (unloaded) motor. 
If those don't work either, then the problem is something else. There are some possibilities but without seeing your wiring and code, these are some things to check:
- if SLEEP is connected instead of ENABLE (this disables the driver right when it should step)
- check if microstep levels match, if not set by the code (this could cause the driver to send steps too fast for the motor to turn)

jonathan-moss, 2019-01-28:
Hi Laurentiu,  

Thanks for getting back to me. Your words 'sending steps too fast' got me thinking.  

I basically have a button set up using the OneButton library, and with a single button click I am using `stepper.rotate(360);` with `setMicrostep(8)` (in fact any microstepping setting works from 1 to 32).  

What I am trying to do is move the motor continuously when the button is held and stop it when the button  is released.  

I have just tried to print out micros from next action like so:
`  unsigned wait_time_micros = stepper.nextAction();`  
`  Serial.println(wait_time_micros);`  

The print out I get is always the same:
`304`  
`304`  
`304`  
`etc.`  

I then tried to print `millis()` and could see it always incremented by 32, so the pulse from `nextAction` was taking too long. There must be some delay in calling long press in the oneButton library.  

I have just tried it with `stepper.move(320);` and as long as the value is greater than microsteps * millis, the motor moves perfectly. I can even set the RPM to whatever I need.

This seems to be the perfect solution for non blocking code. The only problem I see so far is that stepper.move() can't be stopped with stepper.stop() so I will have to think about how to introduce some end stop rules here. (eg. if you set move to a very high number the motor continues to spin after the button is released).

I will run some more tests and do a code cleanup and see how it goes.

Thanks again for the assist.

laurb9, 2019-01-29:
I took a quick look at the OneButton code and I don't see a reason it could be taking too long, as it has no internal loops. So it may have something to do with how it's used, or there are other callbacks that waste time.

The way I imagine you have it set up is you call `startMove()` from the buttonDown callback and `stop()` from buttonUp, with `nextAction()` and OneButton `tick()` in your main loop. If so, I think it should work.

jonathan-moss, 2019-01-29:
Sorry for wasting your time @laurb9 

I set it up exactly as you said and still got slow motor movement, so I tried it by just bypassing oneButton (using a pin pulled high) and the motor spins as it should.

I realised I had a `Serial.println` running while the button was held down. I removed that and everything works exactly as expected.

I had no idea `Serial.println("Button 1 longPress...");` would cause such a delay!

Thank you for your help.

laurb9, 2019-01-30:
Yes, Serial will impact time-sensitive code, try setting it up for the fastest speed if you have no other requirement for it to be slower.

Thanks for updating, I am going to mark this issue closed.

closed 2019-01-30 by laurb9