laurb9/StepperDriver #78: Fix SyncDriver for accelerated move

laurb9, 2019-02-03:
SyncDriver proportional rpm reducing method does not work well for accelerated mode because the timing for acceleration/deceleration phases is not linear.

Update the math to calculate plateau rpm correctly.

laurb9, 2019-02-09:
A little bit of quadratic equations and this is working correctly now, at least with regard to timing calculation (there is still a slowdown somewhere else but it's not related to the sync math).

closed 2019-02-09 by laurb9

jamesdelaneyie, 2019-02-13:
Hi there, 

I was having issues today with getTimeForMove() returning incorrect values when using Accel/Deaccel unless they were about ~2100. Working a-ok on constant speed. Didn't realise v1.2 was out, will upgrade library tomorrow and test, but am was wondering if the above fix might have also fixed this issue I was having?

I also removed startMove from the function so that I could call startRotate on a deg, and then use other functions to work out the time remaining. Seemed to work fine for constant speed but again, not with Accel

laurb9, 2019-02-14:
Yes, it does sound like the problem (issue #39 ) that was fixed in 1.1.4. Try the newest version.

jamesdelaneyie, 2019-02-14:
I did update the library last night but was still having issues. I'll remove all excess code and add here when I'm working on it again. From memory all my setting are like the below, rather low in general 
 
RPM: 10
ACCEL: 50
DEACCEL: 30
MICROSTEPS: 8

Using MP6500 Polulu Driver, Arduino Mega. 

Great library btw, thanks a mil for making it. Should really make a seperate issue here but I updated Multidriver to run 4 motors, worked great. Planning to update to 12 later on. Think I'll run into any issue?

laurb9, 2019-02-15:
With 12 motors you would see a progressive slowdown and a low rpm floor, but at the speeds you are talking about it should still work. I'd be more concerned about having enough GPIO pins.

Consider using constant speed. The motor can start moving directly at 10RPM, unless it has a very large load.

laurb9, 2019-02-22:
Interesting application you have there. Indeed you might need acceleration or if using constant speed, you could try varying the rpm while moving. It'll only get slower when many motors move at the same time.

begin() you should only call once - it sets up the pins. If `setSpeedProfile()` is called while moving in accelerated mode, nothing will change because `startMove()` precalculates the acceleration parameters at the start of the move only. If you call startMove() while moving, it'll try to start again from 0rpm.