laurb9/StepperDriver #84: MOTORS DO NOT WORK ON MICROSTEPS...

LeandroLimaPRO, 2019-12-03:
Firstly I thank the library, very rich.
**DESCRIPTION:**
I am designing a "cnc corexy", I use the following equipment:
1x Arduino one (generic).
1x CNC shield v3.
2x A4988 module
2x Stepper motor NEMA 17 4401
The desired microstep settings are 1:16 (set pins M0 to M2)

**SOFTWARE CONFIGURATION:**
```
#include "A4988.h"
#define STEP_RESO 200 // STEPS FOR RESOLUTION
#define MICRO_STEP 16 // has been tested 1 (partially works) from 2 to 16 (does not work)
#define STEPIN 3 // ARDUINO DOOR TO STEP PIN
#define DIRPIN 6 // ARDUINO DOOR TO DIR PIN
#define ENA 8 // ENABLE PIN
// the rest of the code just sets the settings and commands the motor to rotate 360 ​​clockwise or counterclockwise OR ALL OTHER EXAMPLES AVAILABLE.
```

**PROBLEM:**
Currently I only test the motors, with the available lib examples, but when using the 16 microsteps configuration and also configured in software, the stepper motors have "reverse pole effect", the famous vibrations and step losses. But when used fullstep (1), obviously without pins on M0 to M2, works perfectly. As long as the RPM is below 120.

_Note: 
1. At 1:16 the engine does not run perfectly at any RPM setting.
2. I've tested it with my 3D printer's drives and motors (it's fully operational) - to no avail either. 
3. motors properly calculated and tested on the already informed 3d printer.
_

LeandroLimaPRO, 2019-12-06:
was cabling problem (poor contact)

closed 2019-12-06 by LeandroLimaPRO

laurb9, 2019-12-11:
I would not recommend using this library for a CoreXY CNC machine. The timing is done in the main thread, so any other operation done on the same processor will impact it. Multi-motor sync at higher step rates is not working well partly because of this too.

There's a more mature library called AccelStepper. It has a higher learning curve and a different license that may or may not work for you, but I believe it uses interrupts so it's more suited for high-speed operation. You can use this lib to configure the microstepping in software and the other one for step.

Another option for you use the grbl library on a standalone arduino board. It has all the code to operate a cnc machine and it supports g-code already.