laurb9/StepperDriver #91: Microstep multiplies degrees

hspeedy, 2020-04-18:
I'm using an A4988 driver with all microstep pins connected.
Steps are set to 200. Setting microstep to 1 and calling startrotate with 360 works fine.
Settting microstep to 2 and calling startrotate with 360 results in two turns.
Setting to 4 gives me four turns. The code is taken from master branch.

laurb9, 2020-04-19:
First verify that you are using the A4988 class and not the BasicDriver, and that all the parameters were given in that order:
```c++
#include "A4988.h"
#define MS1 10
#define MS2 11
#define MS3 12
A4988 stepper(MOTOR_STEPS, DIR, STEP, SLEEP, MS1, MS2, MS3);
```
Then try setting the microstep control pins manually - either physically or via digitalWrite in the setup code. The code has been tested with Pololu A4988 boards. You may have a board whose microstep controls aren't actually wired or has other problems.

hspeedy, 2020-04-20:
I'm using A4988 class. The board was purchased from AZ-Delivery. According to datasheet all
microsteps up to 16 are supported. So the board should be fine for this. I will check the pins
this evening and post my results.

hspeedy, 2020-04-20:
omg, my fault :-( I mixed up order for constructor parameters

closed 2020-04-20 by hspeedy