laurb9/StepperDriver #92: Loosing steps during acceleration and deceleration

sigurdild, 2020-04-24:
Is it possible set a cut in and cut out speed during acceleration. I have noticed a step loss and expect it to happen at low rpm during the constant acceleration and deceleration. I can minimise is it to a certain degree by using very high accelerations.

laurb9, 2020-05-16:
Having a minimum speed could be useful, but I do not understand why slower speed is a problem here.

Are you observing the stepper is losing steps, or is the library sending fewer steps ?
A stepper should be able to handle any slow speed since it's moving step by step. There is a ClockStepper example which does just that, move the stepper one step per second.

laurb9, 2020-09-02:
Closing (stale)

closed 2020-09-02 by laurb9