laurb9/StepperDriver #96: Refuses to move high number of steps

joymonkey, 2020-09-16:
**Describe the bug**
I'm having trouble telling my stepper to move a large amount of steps. This sketch successfully moves the motor 22934 steps, but when it is told to move 45868 steps, it will only move 19668 steps (checked using stepper.getStepsCompleted(), but I can also visually tell that the steps were not right).

**To Reproduce**
https://gist.github.com/joymonkey/b95d479af7c2375a7479c3a6775653e7

**Expected behavior**
Motor should move 45868 steps.

**Platform Setup (please complete the following information):**
 - Arduino IDE version 1.8.13
 - Board type : Arduino Nano Every
 - Stepper driver type : Pololu MP6500
 - Wiring : driver has microstep pins MS1 and MS2 set high (1/8th step setting)

**Additional context**
Add any other context about the problem here.

joymonkey, 2020-09-21:
Tested with full steps and confirm same behavior.
Swapped out my 400-steps-per-rotation stepper motor for a more standard 200-steps-per-rotation motor and still seeing the same behavior.
Also did some more tests...
When commanded to move 32767 steps, it moves appropriately.
When commanded to move 32768 steps, it refuses to move at all.
When commanded to move 32769 steps or higher, it moves an incorrect number of steps.

laurb9, 2020-09-21:
Thank you for providing the details and the sample code. The number seems to indicate an issue with 16-bit signed math as you probably surmised, but we use long even on 8-bit platform... will take a look

joymonkey, 2020-09-23:
I replaced the Arduino Nano Every with a knockoff Arduino Nano (common version with old bootloader), and this problem does not seem to persist. So it seems to be a problem particular to the with the Arduino Nano Every.

laurb9, 2020-09-24:
I was unable to reproduce the problem with an Arduino Uno either. I don't have any megaAVR boards unfortunately.

In Arduino the Every board has an option for 328 register emulation, is there a difference when enabling it ?

laurb9, 2020-09-24:
Hmm, I have a suspicion. Can you please add this to the test sketch and let me know if it prints the expected value or not ?

```c++
  long v = 32768;
  Serial.println(abs(v));
```

joymonkey, 2020-09-25:
> Hmm, I have a suspicion. Can you please add this to the test sketch and let me know if it prints the expected value or not ?
> 
> ```c++
>   long v = 32768;
>   Serial.println(abs(v));
> ```

It prints the value as -32768 (negative)

laurb9, 2020-09-25:
Thank you for confirming, I will be creating 1.3.1 to hopefully address this

closed 2020-10-29 by laurb9