laurb9/StepperDriver #98: Allow for Infinite Rotation

irl-ops, 2020-10-15:
**Is your feature request related to a problem? Please describe.**
Eventually the number of degrees of rotation will overflow if you keep adding to it.  

**Describe the solution you'd like**
The option to set a target RPM and direction and the stepper would continue to move until told to stop.  It would be fantastic if you could also optionally use acceleration and deceleration as well.  Keeping track of position in degrees, not number of steps to undo the move, would be nice.  

**Great library!  Thank you for your hard work!**

laurb9, 2020-10-29:
Using non-blocking mode and constant speed, this is possible to do without changes to the library. It's still better to keep track of the position in steps (modulo 200) because being a float value, the angle may drift over a large number of rotations. The internal step counter of the library does not matter even if rolls over, but care may be needed to it does not land on 0.

laurb9, 2021-03-06:
Closing, presumed resolved.

closed 2021-03-06 by laurb9