laurb9/StepperDriver #101: get current position

sharkyenergy, 2020-12-21:
Hello! thanks for the library!

Is there a way to get the current motor position? (while standing still and while moving..)

Ideally as absolute value relative to the starting point...

Thank you

laurb9, 2021-01-03:
Hi, 

The method `getStepsCompleted` can be used to get the number of steps completed since the move started.
This is of course useful mostly in non-blocking mode.

Houri93, 2021-03-22:
I have just had to implement this, modify the library so a long number is set to 0 at the constructor, named it currentPosition.
within nextAction, check if dir_state is positive, add one to currentPosition, otherwise subtract one.

The library implements relative movements of the current location using startMove, I have implemented a method startMoveToPosition which has a long pos input parameter and simply does startMove(pos - currentPos).

also implemented get and set methods for currentPos.

this is done in BasicStepperDriver file.

closed 2021-09-18 by stale[bot]