laurb9/StepperDriver #102: Code Analysis

Houri93, 2021-03-22:
I have been doing a bit of research testing and improving Arduino stepper libraries, yours is the best one so far after improvements, my use case is to drive 6 motors Independently on an Arduino Due at high speeds and acceleration with "async" actions, performance is key in this situation as other tasks are performed on the same processor between nextAction calls so here are my suggestions for others attempting to do the same.

- Implement an absolute position method to move a stepper to an absolute position rather than relative steps by keeping track of the current steps in a counter number which increments or decrements at nextAction method based on dir_state variable.
- combine acceleration and deceleration into one variable, this means that the equations using those two variables can be simplified which causes startMove to become faster.
- in start move, the equations for steps_to_cruise, steps_to_brake, step_pulse and cruise_step_pulse can be moved to setSpeedProfile since all parameters are being revaluated with the same resulting values, i have made more variables and moved those equations to setSpeedProfile, then set the values at startMove.
- I have implemented inversion boolean, direction pins.
- I have removed all calculations for micro stepping, rpm and degrees, everything is dealt with as steps which removes a lot of equations in my situation.
- The first delay in nextAction is removed and replaced with: if (micros() - last_action_end < next_action_interval) return;
- equations in calcStepPulse can be simplified, I'll post in the comments if I am able to do so.

[BasicStepperDriver.zip](https://github.com/laurb9/StepperDriver/files/6180850/BasicStepperDriver.zip)

Houri93, 2021-03-22:
a redundant check for the value of (step_count < steps_to_cruise) while calculating rest and accelerating.

laurb9, 2021-03-30:
Thank you for your feedback and kind words, I am glad you were able to modify the library to suit your needs.

I do have to say that it's not the best library for performance with multiple motors. There are a few others that make use of the hardware timers on supported platforms, which offers better performance for many motors. I think AccelStepper is one of those you might want to give a try.

If your intent is to share your changes with others, I would encourage you to fork the repository, which would create a copy linked to the original that you can then modify in your github account and publish the changes there, because this issue ticket will eventually be closed.

closed 2021-09-26 by stale[bot]