laurb9/StepperDriver #110: Not smooth movement

guestisp, 2022-06-13:
I'm the only one seeing a not smooth movement ? There is a little jerk in the rotation , something a small delay...
The same happens on TMC2208 and TMC2209 (in legacy mode, so they are drop-in replacement for A4988 and similiar)

This is the simple code I use:
```
    this->Stepper.enable();
    this->Stepper.stop();
    this->Stepper.setRPM(50);
    this->Stepper.rotate(3650);
    this->Stepper.disable();
```

laurb9, 2022-06-16:
What processor are you using ?

guestisp, 2022-06-16:
esp32. the stepper code is running in a separated task on a different CPU core. seems that adding priority to the stepper task fix the issue but kills everything else (ie the webserver). now i'm testing with the non blocking code (i did a simple while loop on `nextAction`) and works a little bit better but still slow down when the webserver is sending the webpage

slowing down is not an issue, as long the exact amount of steps asked is done (maybe in a little bit more time). could you comfirm this?

any raccomandations for tmc2209?

laurb9, 2022-06-21:
The movement is based on step counter, so it will send the exact number of pulses requested, which translates most of the time to the same number of motor steps. Reversing direction, moving too fast, too much load etc may cause missed steps.

Some ESP32 versions are single core and some are dual core. Can you check which one is yours ?

guestisp, 2022-06-21:
it's dual

closed 2022-12-23 by stale[bot]