laurb9/StepperDriver #112: Smooth transition between movements in LINEAR_SPEED Profile

vovagorodok, 2022-06-19:
There no posibility to make movemebts where starting/ending RMP is not zero.
For example moves like:
[image]
Proposed in `BasicStepperDriver` to have method like:
```
void move(long steps, float startRPM, float endRPM);
or
void move(long steps, float endRPM); // end remember last/startRPM = endRPM
```
For more motors:
```
void move(long steps1, long steps2, float endRPM1, float endRPM2);
```
It will open possibility to make constructions/chains like:
```
void move(steps_to_x1, steps_to_y1, -1, 4);
void move(steps_to_x2, steps_to_y2, 4, 1);
void move(steps_to_x3, steps_to_y3, 0, 0);
```

Or there is another idea how to me that moves?

laurb9, 2022-06-21:
This is a limitation of this simple library. This library precalculates the expensive acceleration parameters before the move. To change speed on-the-fly it would have to do so between steps, it might take too long for slower chips.

You could try the [AccelStepper library](https://www.arduino.cc/reference/en/libraries/accelstepper/)

laurb9, 2022-06-21:
If the movement is complex and the project can afford an additional 328, you could load [GRBL](https://github.com/gnea/grbl) (the CNC control language that 3D printers also use) and send G-code commands to it over serial from your main processor. It's designed for multi-axis control so you can easily drive even 3 or 4 motors with it.