laurb9/StepperDriver #113: Using two steppers - seems that they only can use a single rpm value!!

rogeda, 2022-09-29:
Hi, i tested two identical steppers with identical drivers (DRV8825 compatible) in a stm32 board and made a code in order to:

1) Each stepper makes 10 turns and then stop;

2) Each stepper moves with a different rpm set (one with 100 rpm and the other with 200 rpm);

Obs: non-blocking code.

Result that i expect: stepper with bigger rpm value stop before stepper with lower rpm;

Real result: both stepper stopped at the same time after making 10 turns.

Seems that it´s only possible to define a unique rpm value for both steppers.

Obs.: The same thing is happening with parameter time inside function "startMove". I tested both steppers with different time values.
The steppers stopped at the same time too.
  

My testing code:

/********************* HERE IS THE CODE *********************************
#include <Arduino.h>
#include <DRV8825.h>

//pin define
#define stepper_ENABLE PC3 //stepper enable pin

//stepper microstep config
#define MICROSTEP 16

#define MOTOR_STEPS 200 //step per-revolution

//steper pin config (dir_pin, step_pin)
DRV8825 stepperX(MOTOR_STEPS, PB9, PC2);
DRV8825 stepperY(MOTOR_STEPS, PB7, PB8);

void setup() {
  //pin mode config
  pinMode(stepper_ENABLE, OUTPUT);
  digitalWrite(stepper_ENABLE, LOW);//enable stepper motor (low to enable high to disable)
  
  
  //stepper config
  stepperX.setSpeedProfile(BasicStepperDriver::CONSTANT_SPEED);
  stepperY.setSpeedProfile(BasicStepperDriver::CONSTANT_SPEED);
  
  //begin with different rpm sets
  stepperX.begin(200,MICROSTEP);
  stepperY.begin(100,MICROSTEP);
 
  
  //move stepper motor for demo
  stepperX.startMove( 10*MOTOR_STEPS * MICROSTEP,0);     //prepare values for 10 turns clockwise
  stepperY.startMove( -10*MOTOR_STEPS * MICROSTEP,0);    //prepare values for 10 turns counter-clockwise
 }

unsigned wait_time_microsX = 0;
unsigned wait_time_microsY = 0;

void loop() {
  //stepper demo
  wait_time_microsX = stepperX.nextAction();
  wait_time_microsY = stepperY.nextAction();

  if ((wait_time_microsX <= 0) && (wait_time_microsY <= 0))
  {
      digitalWrite(stepper_ENABLE, HIGH);  //disable stepper motors
  } 
}

/**********************************************************************

Platform Setup (please complete the following information):
 - VSCode Platform.io with Arduino framework;
 - Board stm32F103RET6 (board 4.2.2 from Creality);
 - Stepper driver type : DRV8825 compatible (Obs.: Microsteps are hardwired in this board for 1:16);

Obs.: I tested both steppers in this exactly same platform setup with a code using funtcion "rotate".
Using rotate(360) with different rpms, worked well as expected!!

laurb9, 2022-10-03:
The `MultiDriver` or `SyncDriver` classes are intended for controlling two or three motors. `MultiDriver` uses the RPM set for each motor, while `SyncDriver` adjusts speeds so they all complete at the same time. The [MultiAxis example](https://github.com/laurb9/StepperDriver/tree/master/examples/MultiAxis) works for both.

The reason the code above does not work as expected is due to how `nextAction` sends a step pulse each time it is called, waiting if needed. As each one is called with each iteration, that guarantees both drivers will emit the same number of pulses in the same time frame, slowed to the lowest RPM.

rogeda, 2022-10-03:
Many thanks for the reply. 
I will try to use MultiDriver / SyncDriver classes.
Congratulations for the good work implementing this Stepper library.

closed 2022-10-03 by rogeda