laurb9/StepperDriver #115: Stepper scaling

Bennisov, 2022-12-19:
Hi!
Im using A4988.h library and want to make a stepper that steps by a set degree value (taken from serial input), and saves records from a lidar. Im using function stepper.move() with steps per rev = 200, RPM = 1 , microsteps =1. What would be the correct way to trasform my input in degrees to value to this fuctions. Thx in advance

D0D0mitWaffe, 2022-12-19:
If i understand you correctly, you can just use the stepper.rotate(double degrees) function. It takes a double value as degrees as a parameter.

Bennisov, 2023-01-03:
Thank you!
I have another issue, I have a code that moves the stepper in a loop, and scans distance from a lidar detector. It takes the ammount of steps as readseries and prints the angle that it covered and distance as print series. It works as intended for most of time, but from time to time it prints nothing, moves for a greater angle (1 step given, moves for 20), skips all steps in between and continues to work properly with correct angle (series: @50 deg@ @51 deg@ and after jumping eg. 20 deg: @71 deg @72 deg@)
i paste the code below, thx in advance

//#include <Arduino.h>
#include "A4988.h" 
#include <Arduino.h>
#include <Wire.h>        // Instantiate the Wire library
#include <TFLI2C.h>

TFLI2C tflI2C;

int Step = 3; //GPIO3 in Arduino UNO --- Step of stepper motor driver
int Dire  = 2; //GPIO2 in Arduino UNO --- Direction of stepper motor driver
int Sleep = 4; //GPIO4 in Arduino UNO --- Control Sleep Mode on A4988
int MS1 = 7; //GPIO7 in Arduino UNO --- MS1 for A4988
int MS2 = 6; //GPIO6 in Arduino UNO --- MS2 for A4988
int MS3 = 5; //GPIO5 in Arduino UNO --- MS3 for A4988

int16_t  tfDist;    // distance in centimeters
int16_t  tfAddr = TFL_DEF_ADR;  // Use this default I2C address

//Motor Specs
const int spr = 200; //Steps per revolution
int RPM = 100; //Motor Speed in revolutions per minute
int Microsteps = 1; //Stepsize (1 for full steps, 2 for half steps, 4 for quarter steps, etc)
 
int sumsteps = 0;

//Providing parameters for motor control
A4988 stepper(spr, Dire, Step, MS1, MS2, MS3);

 char myData[20];

void setup() {
  Serial.begin(9600);
  Serial.setTimeout(400);
  pinMode(Step, OUTPUT); //Step pin as output
  pinMode(Dire,  OUTPUT); //Direcction pin as output
  pinMode(Sleep,  OUTPUT); //Set Sleep OUTPUT Control button as output
  digitalWrite(Step, LOW); // Currently no stepper motor movement
  digitalWrite(Dire, LOW);
  
   
    Wire.begin(); 
  
  // Set target motor RPM to and microstepping setting
  stepper.begin(RPM, Microsteps);
}
 
void loop() {
    digitalWrite(Sleep, HIGH); //A logic high allows normal operation of the A4988 by removing from sleep
  do
  {  
  if( Serial.available()!=0)
  {
    byte m = Serial.readBytesUntil('\n', myData, 20);
    myData[m] = '\0';  //insert null charcater
    int steps = atoi(myData); //converts string to int
    sumsteps += steps;
    sumsteps=sumsteps%400;
    //Serial.println(String(sumsteps));
    
    if(sumsteps<=200) stepper.move(steps);
    else stepper.move(-steps);
    break;
  }
  } while(1);
     
    do
    {   
        if(tflI2C.getData(tfDist, tfAddr))
        {
        String A;
        if(sumsteps<=200)  A = A+String(200*sumsteps/200.); \\poprawic
        else  A = A+String(360*(200-(sumsteps-200))/200.);
        A=A+String('@');
        A=A+String(tfDist);
        Serial.println(A);
        break;
        }
    } while(1);
    delay(100);
}

closed 2023-10-15 by stale[bot]