laurb9/StepperDriver #119: Please add non 2^n microstep support for close loop stepper.

peter58228, 2023-03-26:
When I was using MakerBase servo 42c and set the microstep to 10，this setting   `stepper.begin(300, 10);`  seems not work!

laurb9, 2023-04-04:
I am not familiar with this driver, but the [MakerBase Servo 42c](https://github.com/makerbase-mks/MKS-SERVO42C) repository has some information on it including microstepping which they refer to as "subdivision":

> Subdivision | 1-256(1,2,4,8,16,32,64,128,256 can be set by the key,and others can be set by serial port)

Microstepping works by halving the step multiple times, it would be an odd device that doesn't do it in powers of 2.

Also I have noticed the number of steps in the begin call (300) is an unusual one. Most steppers shave 200 or 400 steps, some have 48 or 64. I might be able to offer more help with more information about is being attempted here.

laurb9, 2024-03-02:
Closing due to no activity, feel free to reopen with new information

closed 2024-03-02 by laurb9